lardemua/atom #1004: Problem with jsf lists

miguelriemoliveira, 2025-01-22:
I was using -jsf (joint selection function and noticed that this input:

    -jsf "lambda x: x in ['gp1_m_j1',  'gp1_m_j2',  'gp1_m_j3',  'gp1_m_j4'  'gp1_m_j5']" \

was giving wrong results. After a while I saw that there is a comma missing between gp1_m_j4 and gp1_m_j5.

Atom should warn about this  and abort.

miguelriemoliveira, 2025-01-22:
This is the relevant piece of code.

    def filterJointsFromDataset(dataset, args):
        """
        Filters some joints to be calibrated from the dataset, using the joint_selection_function
        :param dataset:
        :param args: Makes use of 'joint_selection_function'
        """
    
        if 'joint_selection_function' in args:
            if not args['joint_selection_function'] is None:
                deleted = []
                if dataset['calibration_config']['joints'] is not None:
                    for joint_key in dataset['calibration_config']['joints']:
                        # use the lambda expression jsf
                        if not args['joint_selection_function'](joint_key):
                            deleted.append(joint_key)
    
                for joint_key in deleted:
                    del dataset['calibration_config']['joints'][joint_key]
                    for collection_key in dataset['collections']:
                        del dataset['collections'][collection_key]['joints'][joint_key]
    
                print("Deleted joints for calibration: " + str(deleted))
    
        return dataset

I am not sure we can fix this because the string is used as a lambda function ... not sure how to check it ...

manuelgitgomes, 2025-01-22:
Hello!

The string itself can be analyzed,  verify that each " ' " is followed by a " , ".
A bit brute, but it should work.

miguelriemoliveira, 2025-01-22:
Hum, but there are some ' that are followed by characters, i.e. 'gp1_m_j1

miguelriemoliveira, 2025-01-22:
Also, I think we can also write the function like this:

-jsf 'lambda x: x in ["gp1_m_j1",  "gp1_m_j5"]'

Not sure ...

Kazadhum, 2025-01-22:
Hello @miguelriemoliveira! My suggestion, also a bit brute, is to:

1. remove the spaces from the string;
2. use `split(,)`
3. check if each resulting sub-string has two consecutive "'"

What do you think?

miguelriemoliveira, 2025-01-22:
Hey @manuelgitgomes and @Kazadhum ,

thanks to both. I will use your ideas and try to implement.