lardemua/atom #1007: Inclusion of rihibot in the ATOM repo

Kazadhum, 2025-01-23:
Hello @miguelriemoliveira and @manuelgitgomes!

For my work on IMU data collection, I created the `rihibot` simulated system, which is basically `rihbot` with an IMU on the end-effector. In the branch I'm using for development, I added this system directly to `atom_examples`.

Now, I'm basically ready to create a PR, but a question appeared in my mind. IMU calibration is not yet implemented, only data collection. With this in consideration, do you think I should include `rihibot` in `atom_exmaples` right away, with warnings in the documentation that IMU calibration is currently under development, or do you think its best if I move the system to a different repository before creating the PR?

I'm currently leaning towards the second option as it seems "cleaner" to me, but I'd like to ask your opinions.

manuelgitgomes, 2025-01-23:
Hello!

I think the best solution is to create a branch to develop IMU calibration.
Then you can merge everything afterwards!

Kazadhum, 2025-01-23:
Hi @manuelgitgomes! I thought about that when I created the branch I'm working on but, since IMU calibration means also adding the support for IMUs all across the ATOM pipeline, I thought it would result in a very large PR, which I'd like to avoid...

So I thought maybe it would be best if I split it in two, with the first one being just for implementing IMU data collection as well as continuous data collection... Perhaps I exclude `rihibot` from a first PR and only add it later on?

Kazadhum, 2025-02-05:
Closing this after PR #1010

closed 2025-02-05 by Kazadhum