lardemua/atom #1015: Functions for derivation of TF data

Kazadhum, 2025-03-29:
Hello @miguelriemoliveira and @rarrais!

I am writing a module for ATOM with functions to derive TF data over time to estimate accelerations and velocities.

I'll post my progress in this issue.

Kazadhum, 2025-04-01:
It seems that `calibrate` gets the residuals used for optimization from the `objectiveFunction()` function. This function defines residuals for a given sensor according to its modality. I have added this preemptively:

```
elif sensor['modality'] == 'imu':
    # Derivate expected tfs from dataset
    # Compare with actual imu data
    pass
```

miguelriemoliveira, 2025-04-02:
Hey @Kazadhum ,

sounds fine.

Kazadhum, 2025-04-03:
Hi! While #1016 is still being discussed, I have moved development to a temporary branch, named `imu-calibration-2`, which I will merge onto `imu-calibration` once that PR is solved.

In the meantime, I have created this module: [module](https://github.com/lardemua/atom/blob/imu-calibration-2/atom_calibration/src/atom_calibration/calibration/tf_derivator.py)

It will be built to be a set of functions to be used by `objectiveFunction()` to calculate the derivatives of the translation and rotation of transformations.

I am now correcting my calculations for the angular velocities. In [63d7c0c](https://github.com/lardemua/atom/commit/63d7c0c9ca9f71805e24c9a1ec5c721bf53d5661), I would calculate these velocities by converting the quaternions to euler angles (XYZ convention), fitting these across time to a polynomial curve and then derivating them.

This was returning very wrong angular velocities, so I realized that this calculation is incorrect, as Euler angles are applied sequentially and so the angular velocities cannot be calculated like this.

I am now implementing the following:

1. Plot out the 4 elements of the quaternions along $t$: $\mathbf{q}(t) = [q_w(t), q_x(t), q_y(t), q_z(t)] $;
2. Fit these elements to a polynomial curve;
3. Derivate the four curves to obtain $\frac{d\mathbf{q}}{dt}(t)$
4. Calculate the angular velocities by using (seen [here](https://arxiv.org/pdf/0811.2889)):

$$ \boldsymbol{\omega} = 2 \frac{d\mathbf{q}}{dt} \circ \bar{\mathbf{q}} $$

where $\boldsymbol{\omega} = [0, \omega_x, \omega_y, \omega_y]$, $\bar{\mathbf{q}} = [q_w, -q_x, -q_y, -q_z]$ and $\circ$ refers to quaternion multiplication.

Kazadhum, 2025-04-03:
Just implemented this, but my results are still not very good. I'll investigate further tomorrow, but I believe I might have to filter some IMU data that appear to not fit to a curve super well.

miguelriemoliveira, 2025-04-03:
Hey @Kazadhum ,

How do you know exactly that the results are not good? Are you comparing to the  imu data produced by gazebo?

Kazadhum, 2025-04-03:
Hi! Sorry, I should have explained how I see that the results aren't good.

Basically, I'm comparing the angular velocities I get with the ones acquired by the IMU in the dataset. Since the frame I'm using for derivation is the same as the IMU, the results should be the same.

I'm also plotting $\mathbf{q}(t)$ and the graphs I see lead me to believe that the polynomial fitting isn't great. I don't have access to these plots right now, but I will post them here tomorrow.

Kazadhum, 2025-04-22:
Hi @miguelriemoliveira and @rarrais! I have some updates on the derivation of IMU data!

## Error Calculations

In order to properly see if my derivation method is working properly, I decided to calculate the errors of the derivation process at all collections, producing a table like the ones produced by the ATOM evaluation processes.

So now, for every collection, my code now evaluates the linear acceleration and angular velocities through derivation at the respective timestamp and compares them with those captured by the sensor.

## Current Results

For the results presented below, I used a 2nd degree polynomial to fit the translation data, and used 75 IMU data points (the 75 closest to the timestamp of each collection). Additionally, the IMU linear accelerations include the reaction resulting from gravity. I compensate for this value, using 

$$\overrightarrow{g}=[0, 0, 9.81] \ ms^{-2}$$

Angular velocity calculations are disabled for now. I decided to check if the linear acceleration calculations were working first, since they should be simpler.

I also tried using 3rd degree polynomials and the results were very similar.

My current results look like this:

```
+------------+---------------------+-------------------+
| Collection | E_lin_accel (m/s^2) | E_ang_vel (rad/s) |
+------------+---------------------+-------------------+
|    000     |        0.1449       |        nan        |
|    001     |        0.0731       |        nan        |
|    002     |        0.0715       |        nan        |
|    003     |        0.0673       |        nan        |
|    004     |        0.3213       |        nan        |
|    005     |         0.01        |        nan        |
+------------+---------------------+-------------------+
```

## On why the results might look like this

Theoretically, these errors should be close to zero. There is no "noise" introduced in the world-IMU topological path, so Gazebo should return "perfect" data. These errors are probably introduced when fitting the data to polynomials. 

For collections 1,2,3 and 5, I think these errors are acceptable. Having an error value below $0.1 \ ms^{-2}$ when there are 200 IMU datapoints per second doesn't seem severe.

Let's say I wish to calculate a displacement between two IMU datapoints considering these post-derivation values of linear acceleration. Let's assume an initial velocity $\mathbf{\dot{x}} = \mathbf{0}$, and that the correct value of $\mathbf{a}$ is the zero vector (which in turn means that $\Delta\mathbf{x} = \mathbf{0}$).

If I have an error of $+/- 0.1 \ ms^{-2}$ for the linear acceleration along a given axis, this would translate to the displacement (along that same axis):

$$\Delta x = 0.1 \times \Delta t ^2 + 0 \times \Delta t$$

Given the IMU operates at 200 Hz:

$$\Delta x = 0.1 \times \left(\frac{1}{200}\right)^2$$

$$\Delta x = 2.5 \times 10^{-6} \ m$$

I recognize this is a crude way to judge these errors, but I think it might be an initial indicator of the quality of the results. I will plot out these errors eventually to get a more in-depth look to verify that this "hunch" is correct.

But that's for the collections with smaller errors. Collections 0 and 4 present higher errors than the others. I believe this to be related to the polynomial fitting. Below is a plot of the translation data from the TFs, in the X,Y and Z axis, for collection 0.

[image]

And for collection 4:

[image]

The red line is the polynomial curve the data fit to. 

The fitting seems to be using data corresponding to a transtition from a stationary state to a moving state or vice-versa, which results in a poor fitting. 

I think what I can do to fix this is implement B-spline fitting, instead of polynomial fitting. I also believe this might lead to better results for the other collections as well, though that is just a hunch.

What do you think?

miguelriemoliveira, 2025-04-23:
Hey @Kazadhum ,

thanks for the detailed report. Some questions comments.

> So now, for every collection, my code now evaluates the linear acceleration and angular velocities through derivation at the respective timestamp and compares them with those captured by the sensor

Do you think its enough to compare only for the collections? I imagine you have a lot of data in between ...

This is simulation so the stuff read by the sensor should be the ground truth, is that right?

> the 75 closest to the timestamp of each collection

these are the 75 data points that preceded the current time stamp right? No data is provided for the future, correct?

> I recognize this is a crude way to judge these errors, b

makes sense for me. I do not have an intuition about the importance of velocity values and differences, but when you transform them into error in displacement it makes more sense.

> The red line is the polynomial curve the data fit to.

For reference, I would like to see the same plot for a collection with lower error.

> I think what I can do to fix this is implement B-spline fitting, instead of polynomial fitting. I also believe this might lead to better results for the other collections as well, though that is just a hunch.

If the errors are all from the approximation  of the polynomial fitting wouldn't it suffice to increase the degree of the polynomial to get a better approximation? 

Also, you could reduce the size of the window (also, the size should be parameterized in seconds and then transformed to number of samples). I think its more intuitive that way. For example, if you say that you are using 75 samples I have no opinion, but if you say your window has a duration of 5 seconds I will tell its too large.

Keep up the great work!

Kazadhum, 2025-04-23:
Hi @miguelriemoliveira! Thank you for the questions!

> Do you think its enough to compare only for the collections? I imagine you have a lot of data in between ...

I started by doing this because it was simple to do and there were similar things done in ATOM already. It is to me a good first indicator of the quality of the results. But I do want to get a more detailed look into these errors and their evolution during the derivation of the whole dataset. My next step is to plot out the error for each axis of translation (and keeping its signal instead of using its absolute value) along $t$.

> This is simulation so the stuff read by the sensor should be the ground truth, is that right?

Yes. I even have the gaussian noise set to zero in the IMU sensor settings in the URDF.

> these are the 75 data points that preceded the current time stamp right? No data is provided for the future, correct?

I have done both options. Either I use the past X samples to the timestamp $t$ or I can use the X closest timestamps in either direction of $t$. Since the calibration method is carried out offline, I see no clear issue with it. The results don't seem very different either way.

> For reference, I would like to see the same plot for a collection with lower error.

Of course! Here is one for reference, for another collection that has a lower error:

[image]

I'll include these graphs in the last post too.

> If the errors are all from the approximation of the polynomial fitting wouldn't it suffice to increase the degree of the polynomial to get a better approximation?

Kind of. I think by increasing the degree of the polynomial too much, I think we risk overfitting. Maybe we can get away with 5th degree polynomials but much higher than that and I suspect the errors will start to get worse. Either way, I will be making a table comparing the errors with different degree values.

> Also, you could reduce the size of the window (also, the size should be parameterized in seconds and then transformed to number of samples). I think its more intuitive that way. For example, if you [...]

This makes sense! I'll make the changes!

miguelriemoliveira, 2025-04-23:
Great. Let me know when you have more results.

miguelriemoliveira, 2025-04-29:
Hey @Kazadhum ,

thanks for the analysis.

But your analysis is being done with simulated data, right? In this case the spikes should not exist? 

The derivation should theoretically work when transitioning from static to dynamic or not?

Kazadhum, 2025-04-29:
Hi @miguelriemoliveira!

> But your analysis is being done with simulated data, right? In this case the spikes should not exist?
> 
> The derivation should theoretically work when transitioning from static to dynamic or not?

I don't think we should have spikes of this magnitude (hence my suspicion of a problem), but I would say that it is expected to have some errors. The reason for this is that even in simulation, the messages we receive are not exactly continuous, they just have a very high frequency. The "jump" from discrete-time to continuous-time through curve-fitting will inevitably lead to errors.

If there are points where abrupt changes in accelerations in the IMU data occur and they don't have "special treatment" during curve-fitting, these will be "smoothed over" and that would lead to higher errors. I think this is what is happening.

----

Here is the IMU data plotted out for the same time period:

[image]

Something to keep in mind when looking at these results: due to the difference in frequency between the `/imu` and `/tf` topics, I'm not able to convert these accelerations to the world frame or compensate for gravity while keeping the same number of datapoints. Hence, the acceleration in the Z axis starts at $9.8 \ ms^{-2}$ (the imu's frame's Z axis is aligned with the world's Z axis at the beginning).

There are fluctuations, which can be seen in every axis, but they seem small in magnitude (note that the scale of the X graph is not the same as the Y and Z graphs).

What seems to me like the issue is the steep curves in the Y and Z axis (which, if you look at the error graphs in my last post, are the axis with larger errors. These "jumps" occur at the beginning of movement, so it seems my suspicions are somewhat supported.

If this analysis is correct, then perhaps a possible solution involves algorithmically identifying these "jumps" and excluding them from the curve-fitting and the error calculation.

What do you think?

miguelriemoliveira, 2025-04-29:
Looks like you are right. I am still a bit puzzled why the derivation fails when we have a transition. Integrations work with transitions right? 

I like your approach of identifying these transitions and tackle them differently.

Kazadhum, 2025-05-02:
Hi @miguelriemoliveira and @rarrais! I added a function for identifying these transition points, and now filter them out during the procedure!

Here are the results!

**Before (no filtering):**

[image]

**Now (with filtering):**

[image]

So it seems like the results are far better overall! (Though there are still some outliers visible.)

miguelriemoliveira, 2025-05-02:
It looks much better. So what is the idea? to have this special calibration procedure where you only account for continuous movements?

Kazadhum, 2025-05-02:
Exactly! 

I use curve-fitting to analytically derive the velocities and accelerations afterwards. The issue I was running into is that these transition points make curve-fitting less accurate.

Imagine the example where the position in the X axis is described by:

$$x(t) = 0, t \in [0, 2[$$
$$x(t) = f(t), t \in [2, 3]$$, where $f$ is a generic function

What was happening is that I was trying to use data from the neighborhood of $t=2$ (for instance, $t \in [1.9, 2.1]$) to fit $x(t)$ to a single curve when in theory the movement over that period of time is best described by a piecewise function. This was leading to poor fittings, which in turn led to poor derivations and high errors.

I don't know if this explanation is super clear, maybe I could explain the whole process better in a meeting.

Kazadhum, 2025-05-13:
Hi @miguelriemoliveira and @rarrais!

Last week I have been puzzled over some very bad errors that I was getting when using my derivator on larger datasets. The issue clearly was caused by a bug and I've been trying to catch it. I apologize in advance for not presenting these errors, as I forgot to document them.

So I'm now going back to a more step-by-step approach and heavily testing everything. With professor Miguel's help yesterday, I caught a major bug concerning the orientations in the TFs used for derivation.

To test my script again, I recorded a short dataset which only involves a single translation in the Y axis.

https://github.com/user-attachments/assets/b5dbc2bb-b576-4a51-855b-ce8682c7d169

Here are the results I got:

[image]

As you can see, I've also changed the plotting of the errors so that all of the axis are represented in the same plot, as well as the translation data. I'm also going to add here the linear velocity, so we can really see the whole picture.

I didn't do this for the angular velocity error graph because the orientations are expressed in quaternion form.

Looking at the errors themselves, the first thing that I see is that the errors for angular velocity are very minor (though why the error on the x axis rotation is not close to 0 like in the Y and Z axes is unknown to me). For the errors in linear acceleration, they still grow near the transition points I've talked about in my earlier posts. Perhaps my filtering process is either not working correctly or is insufficient.

More updates to come!

Kazadhum, 2025-05-13:
I've also created a basic script for numerical derivation, instead of my current method of curve-fitting and analytical derivation. It is possible that it will yield better results in these transition points. I just need to apply the fixes there to test.

Kazadhum, 2025-05-13:
> For the errors in linear acceleration, they still grow near the transition points I've talked about in my earlier posts. Perhaps my filtering process is either not working correctly or is insufficient.

After broadening the parameters for filtration of these transition points, we get the following:

[image]

Clearly, this improved the results considerably.

miguelriemoliveira, 2025-05-13:
Heu @Kazadhum ,

I don't get these latest results. We have to discuss them.

Kazadhum, 2025-05-15:
Hi @rarrais! 

I agree, it would be better to synchronize in a call or meeting. There have been a couple of developments this week which I would like to talk about.

I'll send an e-mail to set it up!

Kazadhum, 2025-05-21:
Hi @rarrais!

Thank you for the tip! I'll make sure to check it out and report back!

Kazadhum, 2025-05-22:
Hi @rarrais! I tried changing the `max_step_size` parameter, but it broke MoveIt and the robot can't move...

Here's the error message:

`Controller 'scaled_pos_joint_traj_controller' failed with error PATH_TOLERANCE_VIOLATED: wrist_2_joint path error 0.100056`

So it looks like I'd have to look into controllers to fix it, so it might not be the best path forward. After thinking about it, I still think moving to Webots is better regardless of this, though, since it lets me set the acceleration of a motor directly.

Kazadhum, 2025-05-26:
A small update on my progress with Webots!

After having done the tutorials on its basic usage, I have now been able to create a simple "small car"-style robot with an accelerometer and a gyroscope.

The car has 4 wheels, upon which I force a given torque which translates to an acceleration. This acceleration is then published in the IMU ROS message output.

I replicated this acceleration profile, which we used for testing the Savitsky-Golay filter, with that robot.

[image]

Here is a video of the movement, with the imu data to the left.

https://github.com/user-attachments/assets/2a03361e-9f42-488d-a2a1-32d922a70b2c

Notice how the acceleration is at a positive constant at the beginning and then changes to it's symetrical later on, eventually becoming (close to) zero.

Kazadhum, 2025-05-27:
Hi @miguelriemoliveira! 

Yes, these values do make sense! The graph isn't from Webots -- it's the acceleration profile that we came up with while we were testing the curve-fitting and SG filter last week. I was recreating it in Webots.

But yes, this data does make sense!

miguelriemoliveira, 2025-05-27:
Hey @Kazadhum ,

one suggestion. This issue should be multiple issues. A lot of work here.

Great news. Don't you also need to get the imu data as a ros message?

Kazadhum, 2025-05-30:
Hi @miguelriemoliveira and @rarrais!

Since my last post, I've been cleaning up the script a little bit, as well as reworking the rotation derivation to also use the Savitzky-Golay filter.

Using the same dataset, here are the derivation results for the rotation:

[image]

As expected, since the motion is purely translational and only occurs on the X axis, there is no rotation. I'll record a different bagfile where the robot does rotate to further confirm that this derivation is working as intended.

If everything goes right with this check, I believe it would be good to run a few tests with noise in the TF, to see how (hopefully!) the derivation results become worse when a poor TF estimate is given for the IMU.

Kazadhum, 2025-06-03:
Thank you @rarrais!

I've also ran some experiments with noise and it works as expected! Here are some results! For brevity's sake, I've put the results for translation along X and the rotation along X. The effects are similar for the other axes.

### X Translation:
#### No Noise:
[image]

#### With noise (0.1 translation, 0.1 rotation):
[image]

#### With noise (0.5 translation, 0.5 rotation):
[image]

### X Rotation:
#### No Noise:
[image]

#### With noise (0.1 translation, 0.1 rotation):
[image]

#### With noise (0.5 translation, 0.5 rotation):
[image]

As expected, when we introduce noise to the atomic TF which connects the sensor to its parent frame, the derivation results distance themselves from the IMU data. We can also see that bigger noise values result in a greater error.

This indicates that this error could be used in optimization, which I think should be my next step. What do you think about these results @miguelriemoliveira @rarrais?

Kazadhum, 2025-06-03:
After briefly discussing it with @miguelriemoliveira, we decided that it was better to use the axis-angle rotation formula to derive the angular velocity instead of deriving every element of the rotation matrix.

I've made the change and the results are intact!

Kazadhum, 2025-06-04:
Hello @rarrais!

> In these graphs, shouldn't we see some noise on the IMU data? The yellow line appears very flat to me (might be a scale/perception issue). This leads me to question what explains the variation in the derived results (red line)?

The noise is applied to the fixed transformation between the IMU and its parent frame. So it would be expected that the derivation results would worsen, as the translation and rotation data that is derived does not correspond to where the IMU is actually placed. The noise isn't applied to the IMU data, but the distance between the derivation results and the IMU data (i.e. the derivation error) grows.

Kazadhum, 2025-06-11:
Hello @miguelriemoliveira and @rarrais!

I've completed work on the derivation module, and as such I think it's best to close this issue. If any problem arises later on we can re-open this issue or open another in the case of a bug.

I am now implementing the derivation errors in the ATOM optimization process (see issue #1019).

commit c9bb706d (2025-05-13): feat(tf_derivator): improved error graphs and major bug fix #1015

Fixed a major bug that caused incorrect rotation calculations and joined
plots of the errors in the same graph
closed 2025-06-11 by Kazadhum