lardemua/atom #1019: Implementation of TF derivation errors in ATOM's optimization for IMU calibration

Kazadhum, 2025-06-11:
Hello @miguelriemoliveira and @rarrais!

Continuing my work from #1015 and #1017...

I am now working on implementing the inclusion of the derivation errors from #1015 in the ATOM optimization process, in order to calibrate the extrinsic parameters of IMU sensors.

I will update this issue with my progress as I go along!

Kazadhum, 2025-06-12:
The optimization process runs, but I'm ran into an issue where the errors remain the same before and after optimization.

After some testing, I've determined that this is caused by a bug where the optimized tfs don't replace the old tfs in the objective function. Working on the fix...

Kazadhum, 2025-06-12:
I tried using all of the data in `continuous_data` but I think this is leading to some errors... Will try to use only collections like the rest of ATOM to make it a simpler problem and to better integrate it into ATOM.

Kazadhum, 2025-06-13:
Hi @miguelriemoliveira and @arrais!

After changing to using only the data from each collection (as opposed to using the errors at every datapoint in the dataset), the problem still persisted. 

What was happening was that the optimization process wasn't leading to a significant decrease in the residuals. A variety of bugs was behind the problem, which have now been fixed. Here is a video of the optimization process at work:

https://github.com/user-attachments/assets/c62d02e5-3121-4121-9d2b-a3b954529718

I want to be able to visually show the pose being optimized in RViz but I still need to make some adjustments to the `calibrate` script for the `ros_visualization` flag to work. I'll work on it now.

miguelriemoliveira, 2025-06-13:
yeah. Translation is getting absurdely wrong

Kazadhum, 2025-06-16:
Exactly. After printing the parameters I see that the value for translation along the Z axis is 9m when the GT is 0.3m...
Here are the parameters post-optimization:

```
Parameters:
                               V ini   V calib     V GT    E ini    E fin
base_link-accelerometer_x   -0.00840   0.00098  0.00000  0.00840  0.00098
base_link-accelerometer_y   -0.00522   0.00846  0.00000  0.00522  0.00846
base_link-accelerometer_z    0.02850   9.04529  0.03000  0.00150  9.01529
base_link-accelerometer_r1   0.01005   0.00007  0.00000  0.01005  0.00007
base_link-accelerometer_r2  -0.00995  -0.00003  0.00000  0.00995  0.00003
base_link-accelerometer_r3   0.01005  -0.00811  0.00000  0.01005  0.00811
```

My hypothesis is that this an issue with this robotic system in particular. Since this robot does not move along Z, the optimizer has no data to infer its position along the Z axis. Perhaps if I record a bagfile where the robot acquires acceleration along Z (for example, if it's dropped from a certain height), it could calibrate properly.

I'll test it out.

miguelriemoliveira, 2025-06-16:
> Exactly. After printing the parameters I see that the value for translation along the Z axis is 9m when the GT is 0.3m... Here are the parameters post-optimization:
>
> ``` [...]

This value 9 recall the 9.8 gravity acceleration. Are you sure its nothing to do with it?

miguelriemoliveira, 2025-06-16:
Also, your hyphotesis seems plausible.

Kazadhum, 2025-06-16:
> This value 9 recall the 9.8 gravity acceleration. Are you sure its nothing to do with it?

It doesn't make sense to me why it should be, since this is a translation value and the 9.8 is an acceleration. I'm also correcting for gravity before calculating the error in accelerations. I might be wrong about this, but I think it's more likely that the data I'm feeding the optimizer isn't representative enough for the optimizer to "figure out" the Z translation parameter.

> Also, your hypothesis seems plausible. An imu on the hand of a robotic manipulator?

Yes. It's probably a good idea to do that right now, though I would like to do a quick test to make sure this is the issue.